RAKWireless/WisBlock #9: Reset

HGarza-Ideas, 2021-03-04:
Hi, I have a issue, when I try to start this example, I always have to push the button reset. And i would like that i don't need to push that button to start working. Because it is going to be an autonomous GPS.
Thank you.

beegee-tokyo, 2021-03-05:
Is your WisBlock from the Beta-Starter kit?

The first pre-sale RAK4631 have a problem with the bootloader when using the GPS module.

You need to update the bootloader. We have a tutorial for the [bootloader update](https://github.com/RAKWireless/WisBlock/tree/master/bootloader) with the Segger JLink.

Or, if you have a DAPLink device or our RAKDAP1, you can follow this tutorial about [RAKDAP1](https://docs.rakwireless.com/Product-Categories/Accessories/RAKDAP1-Flash-and-Debug-Tool/Quickstart/#installation). When using DAPLink or RAKDAP1, the command to flash the bootloader is    
```
pyocd flash -t nrf52840 RAK4630_bootloader.hex
```     
The Bootloader hex file is in the [Bootloader section](https://github.com/RAKWireless/WisBlock/tree/master/bootloader) of this repo.

HGarza-Ideas, 2021-03-05:
They don't tell if my rak4631 is a beta-started kit, but it doesn't matter if I am using the GPS module or not, always when I connect the wis-block to my computer it doesn't recognize the dispositive, I always have to press the reset button for my computer to detect it.

beegee-tokyo, 2021-03-06:
That seems to be a different problem.
Does the WisBlock start when you connect the USB cable to a charger or supply it from a battery?
Did you try different USB ports?
Did yu try different USB cables?
Can you try on a different computer?

HGarza-Ideas, 2021-03-08:
I used another USB cable and another USB port, also I tried with another computer. I connected a charger and also I have the same problem.

beegee-tokyo, 2021-03-09:
But is your application on the RAK4631 starting?
For example, if you flash a simple LED blink application, does the software start and the LED's blink?

beegee-tokyo, 2021-03-10:
In the video you still have the GPS module connected.
[image]

Please try **WITHOUT** the GPS module.

I have 5 WisBlock systems running here, none of them has this problem. I am not sure what your problem is.

HGarza-Ideas, 2021-03-10:
Without the GPS module it start immediately. But the problem is that the application that I want to do (GPS tracker) use the GPS module.

beegee-tokyo, 2021-03-11:
In this case I am proposing you follow my [first answer here](https://github.com/RAKWireless/WisBlock/issues/9#issuecomment-791214496} and flash the latest bootloader.

closed 2021-04-12 by beegee-tokyo