RAKWireless/WisBlock #15: Deep Sleep example without LoRa

jpmeijers, 2021-04-14:
I was trying to test Deep Sleep on the Wisblock RAK4631. The only way to get the power usage down to 24uA is to initialise LoRa and LoRaWAN. Is there a way to switch everything off without having to enable the full LoRa stack?

A low power example without LoRa would be very useful.

beegee-tokyo, 2021-04-14:
Without initializing the SX1262 you will not get low power. Because the datasheet of the SX1262 says that the chip goes from Reset to Startup and then into Standby mode. You have to actively set the SX1262 into Sleep mode to get the low power consumption.

SX1261-2 datasheet V1.2 page 60
[image]

closed 2021-04-14 by beegee-tokyo

jpmeijers, 2021-04-14:
Yes that makes sense. Is there a simple oneliner or short piece of code that one can use to put the sx126x into sleep? Rather than having to include the complete lora and lorawan libraries? I'm thinking of just sending the correct commands via spi by hand.

beegee-tokyo, 2021-04-15:
Definitely not a _oneliner_
You would need to call some functions directly from the [sx126x_board.cpp](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/boards/sx126x/sx126x-board.h) and take some parts from [board.cpp](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/boards/mcu/board.cpp) and [spi_board.cpp](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/boards/mcu/nrf52832/spi_board.cpp) and 
1) setup _hwConfig as in the first lines of lora_rak4630_init
2) initialize SPI as in initSPI() and the GPIO's as in SX126xIoInit()
3) put SX126x into sleep
```c++
		SX126xAntSwOff();

		SX126xWriteCommand(RADIO_SET_SLEEP, &sleepConfig.Value, 1);
		SX126xSetOperatingMode(MODE_SLEEP);
```

beegee-tokyo, 2021-04-19:
Two liner to put LoRa transceiver into sleep mode:
```c++
	// We are not using LoRa here
	// But to keep power consumption low we need
	// to initialize the radio
	lora_rak4630_init();
	// And send it to sleep mode
	Radio.Sleep();
```