RAKWireless/WisBlock #21: Add LoRa mesh to examples?

samuk, 2021-05-12:
Now that Meshtastic is working on Wisblock could you add it to examples? 
https://meshtastic.org/docs/hardware/wisBlock-hardware

https://meshtastic.org/docs/developers/protobufs/api

beegee-tokyo, 2021-05-13:
@samuk 
Thank you very much for your contribution. It is nice to see that WisBlock gets attention from other programming platforms now.

But please understand, as Meshtastic and the WisBlock source code is maintained by you, we are afraid that it might cause future trouble if we add the current example into the WisBlock repo. Any change, any improvement you will apply would have to be manually added here.

But we are very excited about WisBlock supported by Meshtastic, so we will add it into our list of examples in the [WisBlock documentation](https://docs.rakwireless.com/Product-Categories/WisBlock/Quickstart).

And if you can provide us with a tutorial how to use WisBlock and Meshtastic, we can add this to our [KnowldegeHub Learn](https://docs.rakwireless.com/Knowledge-Hub/Learn/) section.

samuk, 2021-05-13:
@beegee-tokyo great stuff! Thanks, just to be clear @geeksville is the maintainer of the project. The documentation is being improved currently, I'll get in touch again when it's more complete and we have a tutorial.

beegee-tokyo, 2021-05-13:
@samuk
Great news!
Waiting for your tutorial. Just buzz me here on this issue.

beegee-tokyo, 2021-05-21:
@samuk 

Added Meshtastic to our Quick Start Guide.
[RAK4631 LoRa Mesh via Meshtastic](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK4631/Quickstart/#rak4631-lora-mesh-via-meshtastic)

Thank you for supporting WisBlock.

closed 2021-05-21 by beegee-tokyo

samuk, 2021-05-21:
Great stuff! We now have LoRa mesh on low power hardware :) You were right, back when we were playing with Disaster Radio..

beegee-tokyo, 2021-05-21:
😸
Sometimes I am right.