RAKWireless/WisBlock #25: Document how to upgrade bootloader if BLE OTA fails

jpmeijers, 2021-06-10:
In file
https://github.com/RAKWireless/WisBlock/blob/master/bootloader/RAK4630/new/README.md

I would like to see the alternative bootloader upgrade steps listed:

https://forum.rakwireless.com/t/bootloader-fails-to-upgrade-via-ble/4193/3

https://forum.rakwireless.com/t/bootloader-fails-to-upgrade-via-ble/4193/4

Currently information on the Wisblock is very spread out in different locations: Forum, docs, readmes. This makes it very difficult for users to know where to go looking if they have a problem. I'd suggest using the README.md files as primary source, building the docs from them. If a question is answered on the forum, that answer should be included in the readmes.

beegee-tokyo, 2021-06-10:
@jpmeijers thank you for your feedback.

We are constantly working on improvements of the WisBlock documentation. With more and more modules added to the WisBlock family, documentation will be concentrated in our [Documentation Center](https://docs.rakwireless.com/Product-Categories/WisBlock/), where you can already find the [updated guides](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK4631/Quickstart/#updating-the-bootloader) how to update the bootloader of the RAK4631. 
The WisBlock repository will be the storage for source codes mainly and the forum for discussions about user problems.

closed 2021-06-10 by beegee-tokyo

jpmeijers, 2021-06-10:
I still think it is a bad idea having the source and the documents separated like that. Would be much better having them together.

jpmeijers, 2021-06-10:
And my other main issue is that there are multiple versions of documents laying around. How should a user know which one is the latest? Better to consolidate everything so that there is no misunderstanding.

beegee-tokyo, 2021-06-10:
Having code and documentation separated in Github and Website is a common practice.
As I said before, we are working on it.