RAKWireless/WisBlock #38: Examples for RAK 11300 don't work

fabltd, 2022-01-02:
There are errors with the RAK 11300 docs:

(https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK11300/communications/LoRa/LoRaWAN)

Does not referene the RAK 11300

The example does not work in Platform IO - Errors with printf.

fabltd, 2022-01-02:
On this Page:

(https://docs.rakwireless.com/Product-Categories/WisBlock/RAK11310/Overview/)

Links to Install Ardunio or Platform IO cores are for the wrong device.

beegee-tokyo, 2022-01-03:
It does, here is the initialization for the RAK11300, which is the stamp module on the RAK11310:

https://github.com/RAKWireless/WisBlock/blob/ec390aae2534dd79d671b7c8af77409299b90435/examples/RAK11300/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP/LoRaWAN_OTAA_ABP.ino#L82

beegee-tokyo, 2022-01-03:
> There are errors with the RAK 11300 docs:
> 
> (https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK11300/communications/LoRa/LoRaWAN)
> 
> Does not referene the RAK 11300
> 
> The example does not work in Platform IO - Errors with printf.

It does, here is the initialization for the RAK11300, which is the stamp module on the RAK11310:

https://github.com/RAKWireless/WisBlock/blob/ec390aae2534dd79d671b7c8af77409299b90435/examples/RAK11300/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP/LoRaWAN_OTAA_ABP.ino#L82

beegee-tokyo, 2022-01-03:
> On this Page:
> 
> (https://docs.rakwireless.com/Product-Categories/WisBlock/RAK11310/Overview/)
> 
> Links to Install Ardunio or Platform IO cores are for the wrong device.

Thanks for pointing to this flaw.
Please use this guide:
https://github.com/RAKWireless/RAKwireless-Arduino-BSP-Index

closed 2022-01-03 by beegee-tokyo