RAKWireless/WisBlock #39: Lorawan Example does not work in Platform IO

fabltd, 2022-01-02:
It seems the platform IO implementation does not support printf as used in the adruino example:

`src/main.cpp:22:10: error: 'class arduino::UART' has no member named 'printf'; did you mean 'print'?`

beegee-tokyo, 2022-01-03:
Use this guide to install RAK11300/RAK11310 support for PlatformIO. It fixes the problem with `printf`:

https://github.com/RAKWireless/WisBlock/tree/master/PlatformIO

closed 2022-01-03 by beegee-tokyo

fabltd, 2022-01-03:
This does not work. Please test this.

https://github.com/RAKWireless/WisBlock/tree/master/PlatformIO

beegee-tokyo, 2022-01-04:
What do you mean "This does not work"?
It works just fine:
[image]

fabltd, 2022-01-04:
Hi

I missed the python script.

Yes its working now:

Also this link does not work:

https://downloads.rakwireless.com/3D_File/WisDuo/3D_RAK11300.stp

Please can you send me the stp file?

Do you also have an altium pcb module?

Thanks

Thanks

beegee-tokyo, 2022-01-04:
For the STP file, it was put into the wrong folder. It is here: https://downloads.rakwireless.com/3D_File/WisBlock/3D_RAK11300.stp

We do not have Altium/Eagle/Orcad/Allegro libraries. We only provide the layout recommendations: https://docs.rakwireless.com/Product-Categories/WisDuo/RAK11300-Module/Datasheet/#layout-recommendation

fabltd, 2022-01-04:
Thank you

Also I cannot seem to find a pin map:

The physical pins are listed here:

https://docs.rakwireless.com/Product-Categories/WisDuo/RAK11300-Module/Datasheet/#hardware

However in the Arduino IDE they are referenced as:

WB_IO1 - WB_IO6

WB_A0 and WB_A1

Can you please tell me for example:

GPIO7 =  WB_IO??? etc

Thank you

beegee-tokyo, 2022-01-04:
Yes, sorry, the information is in different places because we started with the WisBlock Core module before we decided to sell the bare stamp module as well.

Here is the assignment: https://github.com/RAKWireless/RAK-RP-Arduino/blob/85e519ed2f8278ac9d6c4e735df796561b5b9e58/variants/WisBlock_RAK11300_Board/pins_arduino.h#L28

fabltd, 2022-01-04:
Hi

Can you confirm this:

So WB_IO1 - GPIO 6  ( Pin 8 on module)

This is not listed as GPIO here??)  Is there an issue with GPIO 6?

https://docs.rakwireless.com/Product-Categories/WisDuo/RAK11300-Module/Datasheet/#hardware

WB_IO1 - GPIO 22  ( PIN 27 on the module ?)

Pin 41 - GP10 27 ( Shows as PIN_A1 in the file)  The data sheet does not
list this as analog PIN? Is there a reason for this?
Pin 42 - GPIO 26 ( Listed as analog pin)
Pin 43 - GPIO 28    (Listed as analog pin)

Thanks

beegee-tokyo, 2022-01-05:
Hi @fabltd 

Maybe this helps, I will ask our documentation team to add this to the RAK11300 datasheet.

| Stamp Module Pin | Stamp Module Pin Name | RP2040 Name | Arduino BSP Assignment | Arduino Class Usage | Remark 
| -- | -- | -- | -- | -- | -- |
| 1 | RESERVED | --- | --- |   | NC |
| 2 | USB_DM | USB- | Serial | Serial |   |
| 3 | USB_DP | USB+ | Serial |   |   |
| 4 | I2C1_SDA | I2C1_SDA/GPIO2 | PIN_WIRE_SDA | Wire |   |
| 5 | I2C1_SCL | I2C1_SCL/GPIO3 | PIN_WIRE_SCL | Wire  |   |
| 6 | UART2_RX | UART0_RX/GPIO1 | SERIAL1_RX | Serial1 |  
| 7 | UART2_TX | UART0_RTX/GPIO0 | SERIAL1_TX | Serial1 |  
| 8 | GPIO6 | GPIO6 | WB_IO1 |   |   |
| 9 | UART1_RX | UART1_RX/GPIO5 | SERIAL2_RX | Serial2 |   |
| 10 | UART1_TX | UART1_TX/GPIO4 | SERIAL2_TX | Serial2 |   |
| 11 | GPIO7 | GPIO7 | WB_IO3 |   |   |
| 12 | GPIO8 | GPIO8 | WB_IO6 |   |   |
| 13 | GPIO9 | GPIO9 | WB_IO5 |   |   |
| 14 | GND | GND | --- |   | GND |
| 15 | RESERVED | --- | --- |   | NC |
| 16 | BOOT | BOOT | --- |   |   |
| 17 | GND | GND | --- |   |   |
| 18 | RESET | RESET | --- |   |   |
| 19 | SWCLK | SWCLK | --- |   |   |
| 20 | SWDIO | SWDIO | --- |   |   |
| 21 | VBAT_SX | --- | --- |   | 3.3V |
| 22 | VBAT_IO_SX | --- | --- |   | 3.3V |
| 23 | GND | GND | --- |   | GND |
| 24 | I2C2_SDA | I2C0_SDA/GPIO21 | PIN_WIRE_SDA1 | Wire1 |   |
| 25 | I2C2_SCL | I2C0_SCL/GPIO20 | PIN_WIRE_SCL1 | Wire1 |   |
| 26 | RESERVED | --- | --- |   | NC |
| 27 | GPIO22 | GPIO22 | WB_IO2 |   |   |
| 28 | GPIO23 | GPIO23 | LED1 |   |   |
| 29 | GPIO24 | GPIO24 | LED2 |   |   |
| 30 | SPI0_CLK | SPI_CLK/GPIO18 | SCK | SPI |   |
| 31 | RESERVED | --- | --- |   | NC |
| 32 | RESERVED | --- | --- |   | NC |
| 33 | SPI0_MISO | SPI_MISO/GPIO16 | MISO | SPI |   |
| 34 | SPI0_MOSI | SPI_MOSI/GPIO19 | MOSI | SPI |   |
| 35 | SPI0_SS | SPI_SS/GPIO17 | SS | SPI |   |
| 36 | GND | GND | --- |   | GND |
| 37 | GND | GND | --- |   | GND |
| 38 | RF_LORA | --- | --- |   | NC |
| 39 | RESERVED | --- | --- |   | NC |
| 40 | GND | GND | --- |   | GND |
| 41 | GPIO27 | ADC1/GPIO27 | WB_A1 |   |   |
| 42 | GPIO26 | ADC0/GPIO26 | WB_A0 |   |   |
| 43 | GPIO28 | GPIO28 | WB_IO4 |   |   |
| 44 | GND | GND | --- |   | GND |
| 45 | DVDD | DVDD | --- |   | 3.3V |
| 46 | AVDD | ADC_AVDD | --- |   | 3.3V |

[RAK11300-Arduino-Pin-Assignment.xlsx](https://github.com/RAKWireless/WisBlock/files/7811795/RAK11300-Arduino-Pin-Assignment.xlsx)