RAKWireless/WisBlock #41: Receive Data

Famosi, 2022-01-14:
How to **receive** data instead of send?

beegee-tokyo, 2022-01-15:
Hi,

In the LoRaWAN examples you receive the data in the callbacks for RX, e.g. in the LoRaWAN_OTAA_ABP example it is in 
```cpp
void lorawan_rx_handler(lmh_app_data_t *app_data)
{
  Serial.printf("LoRa Packet received on port %d, size:%d, rssi:%d, snr:%d, data:%s\n",
          app_data->port, app_data->buffsize, app_data->rssi, app_data->snr, app_data->buffer);
}
```

In LoRaWAN are three node classes:
- Class A (lowest power consumption) - the node receives downlink packets from the Helium Console only after he has sent an uplink packet to the Helium Console
- Class B - not supported
- Class C - (highest power consumption) - the node is always in RX mode and the downlink packets from the Helium Console are received immediately

But Class C is not supported by the Helium Console (at least I cannot find how to set it). 

So you receive a downlink packet from Helium Console only after the the RAK4631 node has sent an uplink packet.
You set the node class in the `lmh_init(&g_lora_callbacks, g_lora_param_init, doOTAA, g_CurrentClass, g_CurrentRegion);` call. The forth parameter `g_CurrentClass` sets the node class. Default is Class A. If you want to change to Class C, you have to do on both the node and in the Helium Console

Famosi, 2022-01-15:
Hi @beegee-tokyo thank you so much for the fast response! :smile: 

If it's possible I have another question for you. I put it here but let me know if it would be better to open a new Issue.

I'm using [this example](https://github.com/RAKWireless/WisBlock/blob/master/examples/RAK4630/solutions/GPS_Tracker/GPS_Tracker.ino) to implement a GPS Tracker but without using the `3AXIS sensor`. Everything works fine but I have two problems:
1. When I upload a new version of the code to the RAK4631 it takes too much time to localize itself. Sometimes 1 hour, sometimes 3 hours, sometime 6 hours. Is it normal? Can you explain me why it takes so long or how to fix it?
2. The RAK4632, when localized, send a packet every 20 seconds. However when I move the RAK (e.g driving around the city) the position is updated rarely (like every minute). What i mean is that for 1 minute sends always the same [LAT,LONG] coordinates.

Thanks again for your time. Really appreciate your work!

beegee-tokyo, 2022-01-16:
@Famosi 

Where are you testing the GPS tracker example code? I am getting a location fix within 12 to 30 seconds when outside of the building. Inside the house I never get a location fix when the device is on my table. Moving it to a (not tinted) window, I can get a location fix within ~40 seconds.

Sending every 20 seconds, when moving is too fast in my opinion. 

And last question, how is your GPS antenna located? The antenna has a white sticker on one side. This side should be oriented towards the sky to get a fast location fix.

Famosi, 2022-01-16:
I was testing it on my table (indoor).... that's should be the reason! :smiling_face_with_tear: 
Now outside the window it takes about 1 min. I didn't test it cause I didn't have a 3V battery to move it outside.

I think you can close this issue, hope it helps someone else.
Thank you for your time!

beegee-tokyo, 2022-01-16:
Will close this now.
If you have more questions, you can open a new issue here or (better, because more people can support) open it on our [Forum](https://forum.rakwireless.com/c/wisblock/67)

closed 2022-01-16 by beegee-tokyo