RAKWireless/WisBlock #42: Link to 1906 Environment Sensor is a 404

mkinney, 2022-01-17:
On the main page: https://github.com/RAKWireless/WisBlock/README.md

the link to the "RAK1906_Environment_BME680" points to https://github.com/RAKWireless/WisBlock/blob/master/examples/common/sensors/RAK1906_Environment_BEM680

which is a 404

Perhaps it should be linked to:
https://github.com/RAKWireless/WisBlock/tree/master/examples/common/sensors/RAK1906_Environment_BME680

Note: the difference between "blob" and "tree".

beegee-tokyo, 2022-01-17:
@mkinney 

Thanks for finding this. We corrected the name of some examples and must have missed that one. 

Will be corrected..

beegee-tokyo, 2022-01-17:
Fixed with https://github.com/RAKWireless/WisBlock/commit/d024ab062b23f041c825ac4f86eab2818628225f

closed 2022-01-17 by beegee-tokyo