RAKWireless/WisBlock #43: use releases for bootloaders?

mkinney, 2022-02-02:
Looking at the download https://github.com/RAKWireless/WisBlock/blob/master/bootloader/RAK4630/Latest/WisCore_RAK4631_Board_Bootloader.zip

from the https://github.com/RAKWireless/WisBlock/tree/master/bootloader/RAK4630 page.

Perhaps that zip file should be pushed to a release?

If the CI cannot build them (due to some internal reason, you can always create a release manually and push the files into it). If you do have the steps, but need help adding them to your CI, share the steps, and I'll see if I can help adding it.

I'm planning on consuming that Bootloader in the meshtastic-flasher tool ( see https://github.com/meshtastic/Meshtastic-gui-installer ) and I'd prefer having a stable endpoint with version. In other words, I'd like to know that bootloader release 1.2.3 has a date of xxx  and SoftDevice version yyy when I query the INFO_UF2.TXT file.

beegee-tokyo, 2022-02-02:
Hi @mkinney 
The bootloader is not compiled here in Github. It is compiled somewhere in China on a local machine. We have the sources here only for reference.
But I can create releases here manually for the latest version.

mkinney, 2022-02-02:
Sources would be great, but not necessary.

That'd be awesome if they were added to a "manual" release. Thank you!

mkinney, 2022-02-02:
Also, please let me know when you get rid of the existing link for that zip as I'll need to change it, too. :-)

mkinney, 2022-02-02:
Wow. That was fast. Ignore my last question about the existing link. I'll change to the release file.

Can you add one more thing?

Can you add the info to the release note? (or at least the "Date"):

```
UF2 Bootloader 0.4.2 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: WisBlock RAK4631 Board
Board-ID: WisBlock-RAK4631-Board
Date: Dec  1 2021
SoftDevice: S140 6.1.1
```

and the INFO_UF2.TXT file, too?

[INFO_UF2.TXT](https://github.com/RAKWireless/WisBlock/files/7984577/INFO_UF2.TXT)

mkinney, 2022-02-02:
Otherwise, that release https://github.com/RAKWireless/WisBlock/releases/tag/0.4.2 looks great!

beegee-tokyo, 2022-02-02:
Ok, I will close this then.

Anyway we try to _**NOT**_ update the bootloader too often. Only if there is a bug.

closed 2022-02-02 by beegee-tokyo