RAKWireless/WisBlock #44: in bootloader mode, the RAK4631 5005 and 19003 have same values

mkinney, 2022-02-02:
The vendor id and the product are the same when both devices are in boot mode. 

Is there a way to distinguish the two devices without installing any big libraries? (currently using python, so a `pip` package would be fine as long as it works on mac, windows, and linux.

I would love to have a way to easily query the device for what model it is. They do have different product identifiers when not in boot mode.

I'm using `lsusb` in linux to query. If on a mac, you can use: `ioreg -p IOUSB`. On windows, `Get-PnpDevice -PresentOnly`. I typically will run those commands to a file before plugging in the device, then plug in the device, and do a "diff".

Let me know you need more info or do not understand my question.

beegee-tokyo, 2022-02-02:
The vendor id and product id is coming from the RAK4631, not from the base board. There is no option to determine which base board is used.

I am running into the same problem and it will get worse as we will release new base boards soon.

mkinney, 2022-02-02:
Is there any way to add the info to the INFO_UF2.TXT file?

beegee-tokyo, 2022-02-02:
Doesn't help either. The UF2 is coming from the RAK4631. If I take it off a RAK5005-O and put it into a RAK19003, the UF2 is still the same.

mkinney, 2022-02-02:
Ok. Is there a python program I can run to query the device? Or a small standalone C program?

beegee-tokyo, 2022-02-02:
Is this the question about I2C devices that I answered in Discord?

If it is about the base boards, there is no way to detect from the Core module.

closed 2022-02-02 by mkinney