RAKWireless/WisBlock #50: Rak11300 - Error while uploading the sketch

Famosi, 2022-03-26:
I'm having the error "An error occurred while uploading the sketch" every time I try to upload a sketch to my Rak11300.
I have "show verbose output during: compilation and upload" but no further info are displayed.

This happens with all the examples and even using an empty "new project".

I've followed this guide https://docs.rakwireless.com/Knowledge-Hub/Learn/Installation-of-Board-Support-Package-in-Arduino-IDE/ and also tried to update the firmware as suggested there.

Any suggestion? What can I do? 

I'm using Arduino IDE but also tried using Platform IO

beegee-tokyo, 2022-03-26:
Did you try to run the [post_install.sh](https://github.com/RAKWireless/RAK-RP-Arduino/blob/main/post_install.sh) on Linux?

Famosi, 2022-03-31:
That fixed my issue... thanks :)

closed 2022-03-31 by Famosi

beegee-tokyo, 2022-03-31:
Sorry for that problem.
We took the BSP from Arduino and I am not sure why the script is not running automatically.

But I am glad it could be solved.