RAKWireless/WisBlock #51: Rak11300 - OTAA join failed!

Famosi, 2022-03-26:
I'm trying to run the `LoRaWAN_OTAA_ABP` example for the `Rak11300` core.

I'm getting the following error:
```
12:32:48.121 -> =====================================
12:32:48.121 -> Welcome to RAK11300 LoRaWan!!!
12:32:48.121 -> Type: OTAA
12:32:48.121 -> Region: EU868
12:32:48.121 -> =====================================
12:33:16.042 -> OTAA join failed!
12:33:16.042 -> Check your EUI's and Keys's!
12:33:16.042 -> Check if a Gateway is in range!
```

`Check your EUI's and Keys's!`: EUI and Keys are correct.
`Check if a Gateway is in range!`: The Gateway is 1meter from the device.

I've registred the device on the [helium console](https://console.helium.com) and I receive 3 Join Request replied with 3 Join Accept.
Here is a screenshot: https://imgur.com/gallery/PExIFE6.

**Note:**
I've also tried the example `LoRaWAN_OTAA_ABP` example for the `Rak4631` core. In this case I have no problem. Everything works fine and I receive packets succesfully.

If you need more info let me know.

beegee-tokyo, 2022-03-26:
I updated the LoRaWAN examples for RAK11310 two days ago. https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK11300/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP

Do you use the changed example? 

There was as well a bug in the SX126x-Arduino library. ==> https://github.com/beegee-tokyo/SX126x-Arduino/releases

Please update to latest version 2.0.11

Famosi, 2022-03-26:
I was using an old version of the example. 
Now the device join succesfully :)

However, it should sends a frame every 20s. But it sends only one time (at the start). 

Trying to find out why...

Thank you so much!

Famosi, 2022-03-31:
I needed to make little changes to your code. In particular to the [`tx_lora_periodic_handler`](https://github.com/RAKWireless/WisBlock/blob/e5afcc05c182e5ad86566575e1a099535efbbdf5/examples/RAK11300/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP/LoRaWAN_OTAA_ABP.ino#L275) function.

Here is my working version:
```
void tx_lora_periodic_handler(void)
{
  send_now = true;
  appTimer.attach(tx_lora_periodic_handler, (std::chrono::microseconds)(LORAWAN_APP_INTERVAL * 1000));
}
```

beegee-tokyo, 2022-04-18:
That is one of the problems with the RP2040 BSP. It needs sometimes such workarounds because it is based on mbed.

Glad you could solve it.

closed 2022-04-18 by beegee-tokyo