RAKWireless/WisBlock #59: Gps Tracker example Problem! "check acc"

danieleberibe, 2022-07-18:
Hi I was trying this code and following all the doc I ran into a break in the code the moment before I print the axis data, I am left with "check acc" in the console and it all jams and the console freezes. help me? Thanks in advance

beegee-tokyo, 2022-07-19:
@danieleberibe 
Can you give a little bit more information? Do you have the log before the crash and can you share it? 

The code works just fine for me here.
[image]

Can the WisBlock connect to a LoRaWAN server?
Does it show **`OTAA Mode, Network Joined!`** in the log?
Is the RAK1904 in the Slot D of the RAK5005-O Base Board?
Is the RAK1910 in the Slot A of the RAK5005-O Base Board?

danieleberibe, 2022-07-19:
Yes, of course then I am using the 2nd gen board and I saw that the 1910 module must be inserted in slot D. If I then insert the 1904 module in A or C the code stops me here (attached image) while if I put it as you have me recommended you don't even join.
[image]

beegee-tokyo, 2022-07-21:
As we are discussing this now in Discord, I close it here.

closed 2022-07-21 by beegee-tokyo