RAKWireless/WisBlock #64: Flashing RAK2305 stuck on synchronization process

hasanabs, 2023-01-01:
I have an issue while trying to flash ESP32 WROVER-E (RAK2305 module). When I am flashing my module, it is stuck in the synchronization process. According to the Espressif Flash tools, this issue is because of these problems; 
[image]
however, I did follow all the instructions. 

This is my wiring (using FTDI serial). 
[image]

And this is the flashing process that stuck.
[image]

Thank you in advanced.

Flasher: https://www.espressif.com/en/support/download/other-tools

beegee-tokyo, 2023-01-02:
Did you try to lower the baudrate? Try with 115200 baud.

I see two green cables on IO0 and GND, is this the same cable to make sure IO0 is pulled to GND?

hasanabs, 2023-01-02:
I tried with several baud rate, including 115200.

Yes, right, the same cable is pull-down for IO0.

hasanabs, 2023-01-03:
It turns out that my USB TTL was not good; thank you for the help.

closed 2023-01-03 by hasanabs

beegee-tokyo, 2023-01-03:
Glad you could solve it.

Have fun with WisBlock and let us know if you encounter other problems.