RAKWireless/WisBlock #66: Decrypted LoRa data appears corrupted?

wedgie32, 2023-06-26:
Hi, 

I'm using this sketch to push data to Chirpstack v4. It seems the data is getting corrupted on the way through? The payload looks okay:

data:"AQfoFWoAAXMuAABExw=="

but when I try and parse it, I get:

data:"èjs.DÇ"

Not sure how to troubleshoot this?

beegee-tokyo, 2023-06-29:
What do you mean with decrypt? The payload _**`AQfoFWoAAXMuAABExw==`**_ is Base64 format and already decrypted.
[Decoding it from Base64](https://cryptii.com/pipes/base64-to-hex) into a byte array results in _**`{0x01, 0x07, 0xe8, 0x15, 0x6a, 0x00, 0x01, 0x73, 0x2e, 0x00, 0x00, 0x44, 0xc7}`**_ and depending on which example you use (I am guessing it is the [Environment Monitoring](https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK4630/solutions/Environment_Monitoring)) this results into (following the [payload description](https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK4630/solutions/Environment_Monitoring#environment-monitoring-over-lorawan)

0x01 ==> Environment data
0x07 0xe8 ==> temperature = 2024 ==> 20.24 degree C
0x15 0x6a ==> humidity = 5486 ==> 54.86 % RH
0x00 0x01 0x73 0x2e ==> barometric pressure = 95022 ==> 950.22 hPa
0x00 0x00 0x44 0xc7 ==> gas resistance = 17607 = 17607 Ohm

beegee-tokyo, 2023-06-29:
Answered in [forum](https://forum.rakwireless.com/t/decoder-for-chirpstack-v4/9873/2)

closed 2023-06-29 by beegee-tokyo

wedgie32, 2023-06-29:
My apologies and thank you for the clear explanation. I've been struggling with some of the LoRa concepts, and the encoding/decoding part was the trickiest. Your explanation has helped me hugely. 👍