RAKWireless/WisBlock #69: LoRa P2P RX and TX Radio init, in examples, passes a numeric value, not a boolean causing interop problems

jvdstoel1, 2023-07-21:
In both LoRa P2p RX and TX examples, one of the passed initialization values should be passed as bool, not numeric, as the example currently shows.  
	// Set Radio RX configuration
	Radio.SetRxConfig(MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
					  LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
					  LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
					  0, true, 0, 0, LORA_IQ_INVERSION_ON, true);
Here, bool crcOn is coded as 0, not bool, as it should be.  This was uncovered during an attempt to communicate with an SX1276 module, and making this correction resolved communications.  The P2P TX example is also incorrect.

void (*SetRxConfig)(RadioModems_t modem, uint32_t bandwidth,
						uint32_t datarate, uint8_t coderate,
						uint32_t bandwidthAfc, uint16_t preambleLen,
						uint16_t symbTimeout, bool fixLen,
						uint8_t payloadLen,
						bool crcOn, bool freqHopOn, uint8_t hopPeriod,
						bool iqInverted, bool rxContinuous);

beegee-tokyo, 2023-07-22:
A bool is an integer with a value of 0 or 1. I don't see a problem here.

These examples are for WisBlock modules. The used library does not work with an SX1267 module anyway, it supports only SX126x transceivers.

closed 2023-07-22 by beegee-tokyo

jvdstoel1, 2023-07-22:
My test case was using the rak library with a rak module running an sx1262, and communicating with a SX1276 running on another device.  The example, as originally coded worked with other rak modules, but not with the SX1276.  Making this correction would be trivial and helpful as I spent at least three hours before recognizing this.