RAKWireless/WisBlock #74: File Not found when I try to patch platformio

Bighoneypot, 2024-02-27:
C:\Users\gabri\.platformio\RAK_PATCH>python ./rak_patch.py
Patching RAK4631
Patched RAK4631
Patching RAK11200
Patched RAK11200
Patching RAK11310
File not found.
Traceback (most recent call last):
  File "C:\Users\gabri\.platformio\RAK_PATCH\rak_patch.py", line 66, in <module>
    copy_files_ext(src_dir1, src_dir2, src_dir3, dest_dir1, dest_dir2, dest_dir3, "RAK11310")
  File "C:\Users\gabri\.platformio\RAK_PATCH\rak_patch.py", line 27, in copy_files_ext
    shutil.copyfile(src_3+"/Print.h", dest_3+"/Print.h")
  File "C:\Users\gabri\AppData\Local\Programs\Python\Python310\lib\shutil.py", line 256, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '../packages/framework-arduino-mbed/cores/arduino/api/Print.h'

beegee-tokyo, 2024-02-28:
For which platform do you want to patch?
Do you have Arduino-MBed BSP installed?

The patch tries to add required files for all Wisblock Core modules, if you need only RAK4631 or RAK11200 you can ignore the message.

baira1044, 2024-03-12:
i am trying to flash a new bootloader and wipe the old from an outdated unit i just bought. having the same issue with file not found. im stuck and i am NOT a coder but im not an idiot i just need some help tryin to set up this developer wisblock kit i bbought that has two boards to make two messaging units

beegee-tokyo, 2024-03-12:
What exactly is your output?
Did you install all three BSP's?
If only one is missing, the error will show, but the patch works for the ones that are installed.
[image]