RAKWireless/WisBlock #75: any tutorials on setting up these oled displays on the rak wisblock

baira1044, 2024-03-12:
im trying to get into the rak wisblock for a p2p messenger but my bootloader is outdated and i cant find any that have an oled display. i bought the developer kit and a few oled displays off ebay but cant find any info on how to properly install them. any help would be greatly appreciated. i am novice to this but am very interested and i dont know how to code but have installed arduino trying to follow these tutorials but need help. dont bash me too hard!

beegee-tokyo, 2024-03-12:
The bootloader has nothing to do with the OLED display.
If you are using a RAK4631, you can simply update the bootloader with the UF2 file following this guide ==> https://github.com/RAKWireless/WisBlock/tree/master/bootloader/RAK4630#update-over-uf2-drive

If you want to extend the P2P code with an OLED display you have to write it by yourself. Examples for both are available, you need to combine them.

baira1044, 2024-03-12:
I know the oled has nothing to do with that, I was just asking two
questions really. But thanks for the link!

SoothingMist, 2024-12-24:
This thread may help with the OLED: https://forum.rakwireless.com/t/rak1921-mounting/5091.