RAKWireless/WisBlock #80: RAK5811 with RAK19001/RAK11310

SoothingMist, 2024-11-25:
The example at https://github.com/RAKWireless/WisBlock/blob/master/examples/RAK11300/IO/RAK5811_0-5V/RAK5811_0-5V.ino is not clear to me on how to read from analog inputs A0 and A1. Have searched, read, and tried as thoroughly as I know how, including forum posts. However, I am unable to tell how to tell how to access these two inputs from the programmer's level.

This matter is important to the transition of my GitHub project in LoRaP2P data acquisition and transmission from Arduino/Dragino to RAKwireless. Would appreciate a more-clear example of how to use RAK5811 to access A0 and A1. It is important that I understand this so I can apply it and explain it to others.

beegee-tokyo, 2024-11-26:
Initialize analog input ==> https://github.com/RAKWireless/WisBlock/blob/bc6d3d5c3edb7c05aeda1a1af98028e50e8f2212/examples/RAK11300/IO/RAK5811_0-5V/RAK5811_0-5V.ino#L38

Read analog input ==> https://github.com/RAKWireless/WisBlock/blob/bc6d3d5c3edb7c05aeda1a1af98028e50e8f2212/examples/RAK11300/IO/RAK5811_0-5V/RAK5811_0-5V.ino#L53

[Quickstart Guide for RAK5811](https://docs.rakwireless.com/product-categories/wisblock/rak5811/quickstart) using A1 of RAK5811

To read from A0 of RAK5811, replace WB_A1 with WB_IO4.

beegee-tokyo, 2024-11-27:
Please check attached example. I changed it to read from both A0 and A1 of the RAK5811.

Test setup:
[image]

USB log output:
[image]

Code:

SoothingMist, 2024-11-28:
Wow !!!  This is great. Many thanks. Attached is the first rough draft of what is going into the documentation of the overall project. Had good experiences with this new code. Have a happy thanksgiving.
[Using_RAK5811.pdf](https://github.com/user-attachments/files/17952204/Using_RAK5811.pdf)

beegee-tokyo, 2024-11-29:
Glad it works for you.
will close this issue now.

closed 2024-11-29 by beegee-tokyo